sn4k3/UVtools #1015: [Bug] UVTools v5.0.9 X64 [Linux] Fails under Ubunt 22.04

kdh1966, 2025-04-08:
### System

```Text
UVtools v5.0.9 X64
Operative system: Ubuntu 22.04.5 LTS X64
Processor: AMD Ryzen 9 5900X 12-Core Processor
Processor cores: 24
Memory RAM: 56.06 / 62.70 GB
Runtime: linux-x64
Framework: .NET 9.0.3
AvaloniaUI: 11.2.6
OpenCV: 4.10.0

Sreens, resolution, working area, usable area:
1: 7680 x 2160 @ 100% (Primary) (On this)
    WA: 7626 x 2133    UA: 7626 x 2133

Path:       UVtools_linux-x64_v5.0.9/
Executable: UVtools_linux-x64_v5.0.9/UVtools
```

### Printer and Slicer

- Printer: 
- Slicer: 

### Description of the bug

Neither the Appimage nor the zip release is working. Both have missing dependencies.
I find this surprising with the Appimage since the purpose of the appimage is to contain all dependencies.

ZIP:
Your system doesn't have the required dependencies in order to run.
Those dependencies are required at libcvextern/OpenCV library.
UVtools is built on top of the OpenCV and therefore cannot run.

Missing dependencies:
UVtools_linux-x64_v5.0.9/libcvextern.so: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.38' not found (required by UVtools_linux-x64_v5.0.9/libcvextern.so)
UVtools_linux-x64_v5.0.9/libcvextern.so: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.38' not found (required by UVtools_linux-x64_v5.0.9/libcvextern.so)
UVtools_linux-x64_v5.0.9/libcvextern.so: /lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.32' not found (required by UVtools_linux-x64_v5.0.9/libcvextern.so)

Please install or build the dependencies in order to run the software.
Check the manual page at 'Requirements' section for help.

Appimage:
Your system doesn't have the required dependencies in order to run.
Those dependencies are required at libcvextern/OpenCV library.
UVtools is built on top of the OpenCV and therefore cannot run.

Missing dependencies:
/tmp/.mount_UVtoolFUWu1Z/usr/bin/libcvextern.so: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.38' not found (required by /tmp/.mount_UVtoolFUWu1Z/usr/bin/libcvextern.so)
/tmp/.mount_UVtoolFUWu1Z/usr/bin/libcvextern.so: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.38' not found (required by /tmp/.mount_UVtoolFUWu1Z/usr/bin/libcvextern.so)
/tmp/.mount_UVtoolFUWu1Z/usr/bin/libcvextern.so: /lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.32' not found (required by /tmp/.mount_UVtoolFUWu1Z/usr/bin/libcvextern.so)

Please install or build the dependencies in order to run the software.
Check the manual page at 'Requirements' section for help.

### How to reproduce

Fails immediately when launching under Ubuntu 22.04
Launch UVtools_linux-x64_v5.0.9.AppImage
Or
Launch UVtools from Linux zip

sn4k3, 2025-04-08:
See here: https://github.com/sn4k3/UVtools/issues/1010

kdh1966, 2025-04-08:
> See here: [#1010](https://github.com/sn4k3/UVtools/issues/1010)

Thanks.
Ubunu 22.04 isn't considered an older operating system. It is a LTS with end of life in 2027.

sn4k3, 2025-04-08:
The owner of library don't support it, is upgrading OS out of question?

rkern, 2025-04-13:
FWIW, I'm also running into this, and yes, upgrading my OS is not an option for me. I'm using PopOS!, an Ubuntu derivative that manages NVIDIA/CUDA drivers nicely (critical for me), but they are lagging on supporting 24.04. If worse comes to worse, I'll stick with UVTools 5.0.5, the last version that runs on my OS.

It looks like there are [AppImage builders](https://appimage-builder.readthedocs.io/en/latest/faq.html#what-makes-appimage-builder-different) out there that bundle glibc for this reason, but I have no direct experience with them.

sn4k3, 2025-04-13:
Ok I will compile it on a lower version to support older OS

rkern, 2025-04-13:
Thank you! I sincerely appreciate it.

smogshaet, 2025-04-20:
I believe I might a solution to your problem. Check it out here [click](https://forreststonesolutions.com/robots/)

sn4k3, 2025-04-22:
Check the last version

rkern, 2025-04-22:
It works for me! Thank you very much!

closed 2025-04-22 by sn4k3