sn4k3/UVtools #1019: [Bug] Pixel Editor Drawing Cursor Preview not Visible

glenjimen, 2025-04-21:
### System

```Text
UVtools v5.0.9 X64
Operative system: Linux Mint 22.1 X64
Processor: AMD Ryzen 7 1800X Eight-Core Processor
Processor cores: 16
Memory RAM: 11.37 / 31.27 GB
Runtime: linux-x64
Framework: .NET 9.0.3
AvaloniaUI: 11.2.6
OpenCV: 4.10.0

Sreens, resolution, working area, usable area:
1: 1920 x 1080 @ 100% (Primary) (On this)
    WA: 1920 x 1040    UA: 1920 x 1040

Path:       /home/glen/Software/UVtools_linux-x64_v5.0.9/
Executable: /home/glen/Software/UVtools_linux-x64_v5.0.9/UVtools
Loaded file: BeetLoad12 - Level Scaled.ctb [Version: 4] [Class: CTBEncryptedFile]
```

### Printer and Slicer

- Printer: Elegoo Saturn 2
- Slicer: Lychee

### Description of the bug

When using the pixel editor the user holds shift then left / right clicks to add / remove pixels.  When holding shift UV tools shows a blue preview of the size & shape of the drawing cursor.  This cursor showed on Win10, but does not show on Linux Mint 22.  

### How to reproduce

Open a file, and go to the Pixel Editor.   Choose any brush shape or size and hold shift with your mouse over the drawing area.  There is no blue-ish preview.  

### Files

This issue is present on the default UV Tools logo file at startup.

sn4k3, 2025-04-21:
Can you make a small video with a phone and attach here?
I'm unable to reproduce this under a VM, it can be the framework, can you try the last version 4 to confirm?

glenjimen, 2025-04-21:
Wow, thank you for the amazingly quick reply!
I made a quick video.  In the pixel editor I'm holding shift the whole time I'm placing squares.

https://github.com/user-attachments/assets/3ed9559d-0aa2-47e6-9429-e47988f27339

I'm using UVTools 5.0.9 .  What are you asking me to update?  Part of the framework?

sn4k3, 2025-04-21:
Ok, it looks like a problem in framework, the code that set cursor is handed by framework.

> I'm using UVTools 5.0.9 . What are you asking me to update? Part of the framework?

No, downgrade uvtools to the latest v4...

glenjimen, 2025-04-21:
Gotcha, I just tried UVtools_linux-x64_v4.4.3 and got the same results, no preview cursor shape.

sn4k3, 2025-04-21:
I never used UVtools on a native linux to know if this worked at anytime, maybe it never worked due lack of implementation of custom cursors on linux.
I will check.

glenjimen, 2025-04-22:
> When you hold shift, at least the cursor pointer changes to other shape?

The mouse cursor changes to a sort of square shape yes, but that is the pan symbol.  You can see better in this video.  The mouse cursor is the same if I choose a square or circular brush.  
https://github.com/user-attachments/assets/948eeb49-d398-4afb-aa06-d697e2f11fd8

sn4k3, 2025-04-22:
Ok the shift seems to not work properly as you lack of the tooltip.
I was able to reproduce on VM.
What fixed: Hold Shift, click and release Ctrl while holding shift, the brush and tooltip now shows. Can you check if it also works for you that way?

glenjimen, 2025-04-22:
Aha!  Yes, that does bring back the brush preview!  Of course, each time I release shift I need to tap ctrl again, but that's okay.  Thank you for finding this work around!

sn4k3, 2025-04-22:
Will fix this on next release. Strange thing is that worked like that since forever and no one complained xD

sn4k3, 2025-04-22:
Check the last release.

sn4k3, 2025-04-22:
Thanks, 4 years after the bug is fixed :D

closed 2025-04-22 by sn4k3