sn4k3/UVtools #1021: [Bug] macOS x64 (run on M1 Pro) version randomly freezes when opening a file

b005t3r, 2025-04-23:
### System

```Text
UVtools v5.1.0 X64
Operative system: Darwin 23.2.0 Darwin Kernel Version 23.2.0: Wed Nov 15 21:53:18 PST 2023; root:xnu-10002.61.3~2/RELEASE_ARM64_T6000 Arm64
Processor: Apple M1 Pro
Processor cores: 10
Memory RAM: 10.24 / 21.81 GB
Runtime: osx-x64
Framework: .NET 9.0.4
AvaloniaUI: 11.3.0
OpenCV: 4.10.0

Sreens, resolution, working area, usable area:
1: 1920 x 1080 @ 100% (Primary) (On this)
    WA: 1920 x 997    UA: 1920 x 997

Path:       /private/var/folders/q1/dfcmw2vd76j3vt63bv6_89_40000gn/T/AppTranslocation/C0A9FAD8-6931-472E-9B65-8BB23C798BC6/d/UVtools.app/Contents/MacOS/
Executable: /private/var/folders/q1/dfcmw2vd76j3vt63bv6_89_40000gn/T/AppTranslocation/C0A9FAD8-6931-472E-9B65-8BB23C798BC6/d/UVtools.app/Contents/MacOS/UVtools
Loaded file: UVtools_demo_file.sl1s [Version: 0] [Class: SL1File]
```

### Printer and Slicer

- Printer: Anycubic Mono 4 Ultra
- Slicer: Lychee 7.3.100

### Description of the bug

The app freezes when opening a file. It happens randomly. Once I kill it and do the same thing again, it sometimes works (sometimes I need to try multiple times).

### How to reproduce

1. Open UVTools.
2. Drag adn drop the file.
3. Wait.
4. The progress bar stops, the app freezes (randomly).
5. Kill the app.
6. 6Repeat from 1. and hope for the best.

sn4k3, 2025-04-23:
Downgrade UVtools fixes the problem?

b005t3r, 2025-04-23:
Nope, the previous release has exactly the same issue. However, I changed this today:

[image]

and haven't had that issue yet, so maybe that fixes it?

sn4k3, 2025-04-23:
A lower limit of 16gb is too high. Make it something 4g or 6gb and test.

b005t3r, 2025-04-26:
Upping the limit to 16 GB actually helped :) But what also fixed the issue was installing the M1 version via your script, I no longer see that issue with the M1 build.

closed 2025-04-26 by b005t3r

sn4k3, 2025-04-26:
Well, using demanding app under emulation is never a good thing. native m1 build exists since v3