sn4k3/UVtools #1022: [Bug] PCB Exposure - Gerber import is "clipped" on the Vertical Axis

DanielHaroldLane, 2025-04-24:
### System

```Text
UVtools v5.1.0 X64
Operative system: Microsoft Windows 10.0.22631 X64
Processor: 12th Gen Intel(R) Core(TM) i5-12600K
Processor cores: 16
Memory RAM: 14.30 / 31.79 GB
Runtime: win-x64
Framework: .NET 9.0.4
AvaloniaUI: 11.3.0
OpenCV: 4.10.0

Sreens, resolution, working area, usable area:
1: 2560 x 1440 @ 100% (Primary) (On this)
    WA: 2560 x 1400    UA: 2560 x 1400
2: 1920 x 1080 @ 100%
    WA: 1920 x 1040    UA: 1920 x 1040
3: 1920 x 1080 @ 100%
    WA: 1920 x 1040    UA: 1920 x 1040

Path:       C:\Program Files\UVtools\
Executable: C:\Program Files\UVtools\UVtools.exe
Loaded file: PCB.goo [Version: 0] [Class: GooFile]
```

### Printer and Slicer

- Printer: Elegoo Saturn 4 Ultra
- Slicer: Chitubox (Used to create a "dummy" .goo file.

### Description of the bug

When importing a Gerber file generated using KiCad 9, the gerber's vertical axis is clipped when importing using "Tools > PCB Exposure". This causes some of the PCB Artwork/geometry to be missing.

[image]

[image]

### How to reproduce

Repro Steps:

For reference, the KiCad 9 project used for this can be found [here](https://github.com/DanielHaroldLane/SOP-44-DIP-40-Adapter)

Settings for the gerber export/plot were as follows
[image]

1. Download [gerber.zip](https://github.com/user-attachments/files/19884496/gerber.zip).
2. Extract the two Gerber files, "BIOS Adapter-F_Cu.gbr" and "BIOS Adapter-PTH.drl" from gerber.zip.
3. Use Chitubox to generate a dummy sliced file and save as *.goo format. For convenience, feel free to use [PCB.zip](https://github.com/user-attachments/files/19884534/PCB.zip).
4. Open UVTools and open the extracted PCB.goo file (File > Open > PCB.goo).
5. Navigate to Tools > PCB Exposure.
6. Under the PCB exposure section in the dialog popup, select "Add" and select both "BIOS Adapter-F_Cu.gbr" and "BIOS Adapter-PTH.drl".
7. Select "BIOS Adapter-F_Cu.gbr" from the file preview list under the "PCB exposure pane" within the currently open dialog. [image]
9. Notice that the preview pane displays the Gerber file with a chunk of the vertical axis clipped/missing.
10. Select "Create one layer per gerber file".
11. Click the "PCB exposure" button at the bottom of the dialog.
12. Notice that the imported Gerber image layer is missing some of the vertical axis.

[image]

Anyway, just as an aside, I really love this feature of UVTools and am glad it exists, you did an excellent job with it.

sn4k3, 2025-04-24:
Can you highlight wheres the missing part? I can't find the difference between your red and white image.

DanielHaroldLane, 2025-04-25:
Oh interesting, that does make sense, brigado Tiago!

closed 2025-04-25 by DanielHaroldLane