sn4k3/UVtools #1027: [FEATURE REQUEST] Support Editing and Repackaging of Extracted pwsz-Based Files

ServerOnTheMove, 2025-05-16:
### Is your feature request related to a problem?

Yes. When slicing for the Anycubic Photon M5s using Lychee Slicer, the generated .pm5s files include two problematic header values:

ResinType = 22 – This incorrectly identifies the resin as “high-speed,” which:

Triggers a warning on the printer that must be manually acknowledged before each print.

Causes the printer to adjust print behaviour based on incorrect resin assumptions.

Cannot currently be corrected, as there is no way to edit or repackage the .pm5s file after extraction.

Incorrect machine name – Lychee sets the machine name to "Photon Mono 5s" instead of "Anycubic Photon Mono 5s", which:

Prevents the sliced file from being recognised or imported into Anycubic Workbench.

Limits cross-tool compatibility and breaks intended workflows.

These issues create unnecessary friction for users trying to automate or customise their slicing and printing workflows. While UVTools can extract .pm5s files (which is incredibly useful), there is currently no way to modify and repackage the file with corrected header values.

### Describe the solution you'd like

I’d like UVTools to support editing and repackaging .pm5s files for the Anycubic Photon M5s, specifically enabling:

- Editing of extracted file metadata, such as:
ResinType (e.g. update from 22 to 21)
Machine name (e.g. correct "Photon Mono 5s" to "Anycubic Photon Mono 5s")

-  Repackaging of extracted folders into a valid .pm5s file, suitable for direct upload to the printer or import into Anycubic Workbench.

- Optional Configuration.ini replacement after extraction (e.g. allow drag/drop or file selection to overwrite the original before repackaging).

- Validation checks during packaging to ensure the output conforms to expected structure and naming required by the printer.

Ideally, this functionality could be offered via:

A GUI workflow (e.g. “Open folder → Update metadata → Package to .pm5s”)

Or a CLI command (for automation pipelines or power users)

This would enable users to correct slicer-generated inaccuracies without needing to modify or reverse-engineer the binary format manually.

### Files

[Enhancement Examples.zip](https://github.com/user-attachments/files/20259945/Enhancement.Examples.zip)

sn4k3, 2025-05-17:
You can do this without exit UVtools, via scripting. Which also work via Cmd.
Easiest way is File -> Terminal.

```
((AnycubicFile)SlicerFile).HeaderSettings.ResinType = 21;
SlicerFile.MachineName= "Anycubic Photon Mono 5s";
```

ServerOnTheMove, 2025-05-17:
> You can do this without exit UVtools, via scripting. Which also work via Cmd. Easiest way is File -> Terminal, send:
> 
> ((AnycubicFile)SlicerFile).HeaderSettings.ResinType = 21;
> SlicerFile.MachineName = "Anycubic Photon Mono 5s";
> Save file and its done.
> 
> However you should contact lychee in order to fix this.

Thanks you - if it does what I think it does, that should be all I need.
I will give it a go and I assume I can do this for every parameter.
Is my (other) assumption correct that this will update the file in memory and then the UI save function will save it into the print file?

Thanks,
Dante

ServerOnTheMove, 2025-05-17:
Thank you, this worked great. Sky is the limit now.
Will close the request as this meets my required in full.

closed 2025-05-17 by ServerOnTheMove

sn4k3, 2025-05-17:
> Is my (other) assumption correct that this will update the file in memory and then the UI save function will save it into the print file?

Yes