sn4k3/UVtools #1028: [Bug] Missing printer Elegoo Saturn 4

JackDesBwa, 2025-05-21:
### System

```Text
UVtools v5.1.1 X64
```

### Printer and Slicer

- Printer: Elegoo Saturn 4
- Slicer: PrusaSlicer

### Description of the bug

When installing a profile into PrusaSlicer, the Elegoo Saturn 4 is not available

### How to reproduce

Open Help>install profiles into PrusaSlicer
The printer is not in the list

JackDesBwa, 2025-05-21:
It is what I did initially, but it did not work.
By investigating, I finally saw that it is because the release system is not the same on the two machines.

[image]

That's why I made this printer profile. Today, I noticed that I forgot to share it.
I successfully printed several pieces, more than I initially planned to wait before considering the configuration worth to share.

sn4k3, 2025-05-22:
> It is what I did initially, but it did not work.

Because you forget to rename the printer, UVtools have hardcode for "4 Ultra" machine names.

sn4k3, 2025-05-29:
Added

closed 2025-05-29 by sn4k3