sn4k3/UVtools #1030: [FEATURE REQUEST] UVtoolsCmd: Creating a new file from a folder of images OR replacing an existing file's contents with images

b005t3r, 2025-05-22:
### Is your feature request related to a problem?

I'd like to export the contents of a slicer file to a folder with UVtoolsCmd (currently possible), process those images in my own software and then import back the changes to the original file, copy or a new file created with the processed images.

### Describe the solution you'd like

A new option for UVtoolsCmd. Either:

impoting layers to an existing file, replacing all of the existing layers and removing the excess layers
OR
creating a new file based on an existing file (to copy the settings from) and a folder with images to copy the layer contents.

sn4k3, 2025-05-22:
The UVtoolsCmd can run all operations that the UI can. As so you can export all the images from a file and import it on the other.
However that is a extra step, you can pass the export step and import directly from one file to another.
Looks for `UVtoolsCmd run` command
You will need to prepare a xml file with the operation, which you can get from UI, going to layer actions -> Import layers. Configure as you want, click the menu option (bottom left) and click export settings. Then you can pass the .uvtop file to UVtoolsCmd.

As XML indicate a list of files, you need to manipulate that file path before passing the uvtop file, you can do this easy via PowerShell if on Windows.

b005t3r, 2025-05-22:
> However that is a extra step, you can skip the export step and import directly from one file to another.

The thing is I need the exported images for processing, but I don't see a way to export the images using the command line version. And in the UI I don't see that bottom-left menu button for the Export Layer option:

[image]

Is there a way to create that?

sn4k3, 2025-05-22:
I might have excluded the menu settings on that operation. But is possible anyway:

`UVtoolsCmd run '.\72 spoke 3 bar-apw-3_0 (Anycybic Workbench).pm5s' OperationLayerExportImage -p OutputFolder=MyExportToFolder`

b005t3r, 2025-05-22:
It works, thank you!

sn4k3, 2025-05-22:
UVtoolsCmd also let you run script files, which means, you can make a one file script to automate a ton of works. Including exporting, processing and repack in one go. (C#)

sn4k3, 2025-06-12:
Code completion should work.
Looks like you using Visual Code? Have you installed the C# dev kit extension?
Also to have the project references you should open the folder where `UVtools.ScriptSample.csproj` file is together with scripts.

If you want easy IDE use Jetbrains Rider, is now free and much better than VS code for C#