sn4k3/UVtools #1036: [Crash] System.AggregateException: A Task's exception(s) were not observed either by Waiting on the Task or accessing its Exception property. As a result, the unobserved exception was rethrown by the finalizer thread. (Your system memory RAM hit the limit

scarfcat, 2025-06-13:
### System

```Text
UVtools v5.0.8 X64
Operative system: Manjaro Linux X64
Processor: AMD Ryzen 9 5900X 12-Core Processor
Processor cores: 24
Memory RAM: 47.31 / 62.70 GB
Runtime: linux-x64
Framework: .NET 9.0.2
AvaloniaUI: 11.2.5
OpenCV: 4.10.0

Path:       /tmp/.mount_uvtoolQ2Buqn/usr/bin/
Executable: /tmp/.mount_uvtoolQ2Buqn/usr/bin/UVtools
```

### Printer and Slicer

- Printer: Phrozen Mighty Revo 14k
- Slicer: Lychee

### Description of the bug

The application crashes saying that there's no memory left, but in system monitor it doesn't even get close to 60 GB..?

```
System.AggregateException: A Task's exception(s) were not observed either by Waiting on the Task or accessing its Exception property. As a result, the unobserved exception was rethrown by the finalizer thread. (Your system memory RAM hit the limit of 61.65GB from a total of 62.7GB.  Available: 1.05GB.  UVtools: 31.93GB (50.93%).
The program crashed on purpose due the impossibility to pause or cancel the running operation, this was to relief pressure and ensure the system stability.

Monitor your memory RAM under the system tools and re-run the operation to check the usages and find the cause.
You may need to stop other processes or increase your memory RAM in order to deal with huge files and/or heavy operations.

Currently you have configured a lower limit of 2GB of available memory RAM for the operations.)
 ---> System.InsufficientMemoryException: Your system memory RAM hit the limit of 61.65GB from a total of 62.7GB.  Available: 1.05GB.  UVtools: 31.93GB (50.93%).
The program crashed on purpose due the impossibility to pause or cancel the running operation, this was to relief pressure and ensure the system stability.

Monitor your memory RAM under the system tools and re-run the operation to check the usages and find the cause.
You may need to stop other processes or increase your memory RAM in order to deal with huge files and/or heavy operations.

Currently you have configured a lower limit of 2GB of available memory RAM for the operations.
   at UVtools.UI.MainWindow.<>c__DisplayClass57_0.<RamUsageTimerOnElapsed>b__0() in UVtools.UI/MainWindow.axaml.cs:line 519
   at Avalonia.Threading.DispatcherOperation.InvokeCore()
   --- End of inner exception stack trace ---
Category: Task
File: V24_3.prz  [Version: 0] [Class: GooFile]

Machine date time: 6/14/2025 12:23:27 AM
    UTC date time: 6/13/2025 10:23:27 PM
```

### How to reproduce

Trying to repair/find issues on a file sliced to a Mighty Revo 14k

scarfcat, 2025-06-14:
So in this specific case it must've allocated like 15 GB in less than a second so it's undetectable in system monitor?

Thing is, it was giving me the message and pausing while I was scanning for issues, but my system was at around 30 GB ram at that time, meaning it allocated 30 GB each time while scanning? Since my system monitor didn't even show me go past 30 GB at any time

sn4k3, 2025-06-14:
At forced shutdown time you were with:

> Total RAM: 62.7GB.
> Available (Free): 1.05GB.
> UVtools: 31.93GB (50.93%).

If you or UVtools pause a running operation is normal that RAM reliefs, but if you monitor while processing you can see spikes, the larger the spikes the larger the allocations and deallocations (Heavier operation)
With resolutions going up RAM is no joke, you will need a ton of it. 
You can disable this RAM limiter and rely on SWAP, which on windows work very well. On Linux my experience is that worse handled, but you can try.
This limiter was introduced due this request: https://github.com/sn4k3/UVtools/issues/953

If you want to relief the workload you can simply assign less cores/workers, go to setting and set a lower value of parallel tasks. 
Each task will allocate at least 1 bitmap, which already requires a decompression from memory and then manipulate the bitmap which creates sub-bitmaps, a single task depending on the operation can allocate many bitmaps which are your layer image, that multiplied by your core count is a ton of RAM, the easy days of RAM in FullHD are gone.

As you are hitting RAM depletion I really recommend tune the value to relief RAM, your system without RAM and a program hungry for it can never lead to good things.

[image]

scarfcat, 2025-06-14:
> Total RAM: 62.7GB.
Available (Free): 1.05GB.
UVtools: 31.93GB (50.93%).

But actually, I had nothing else running in the background that would've consumed 50% RAM

I had some other stuff running, like Firefox, that consumed a few GBs at max, but not 30 GB.

closed 2025-06-14 by scarfcat

sn4k3, 2025-06-14:
UVtools get the stats from system.
With UVtools closed see how much ram your system reports, open UVtools and sum previous value with the value UVtools says is taking and compare to what system shows.

scarfcat, 2025-06-16:
At the time of this report, I was checking my system RAM usage. The other things consumed 10 GB, and UVTools crashed when it reached 30 GB (around 40 GB total)

sn4k3, 2025-06-16:
It can happen that what system reports is not the real value. Systems can use RAM up to a value and only start optimize when required. Because optimization/defrag is slow. If you believe that you still have headroom, you can disable the limiter, system will try to relief RAM when near limits. 
UVtools also have a function, under Help menu that forces the RAM optimization of the program, however its never required to call that manually.