sn4k3/UVtools #1038: [Bug] PCB Exposure tool generates unprintable file

felipetrentin, 2025-06-14:
### System

```Text
UVtools v5.1.2 X64
Operative system: Microsoft Windows 10.0.19045 X64
Processor: AMD Ryzen 5 3600 6-Core Processor              
Processor cores: 12
Memory RAM: 7.65 / 15.93 GB
Runtime: win-x64
Framework: .NET 9.0.5
AvaloniaUI: 11.3.0
OpenCV: 4.10.0

Sreens, resolution, working area, usable area:
1: 1920 x 1080 @ 100% (Primary) (On this)
    WA: 1920 x 1040    UA: 1920 x 1040

Path:       C:\Program Files\UVtools\
Executable: C:\Program Files\UVtools\UVtools.exe
Loaded file: pov_pcb.cxdlp [Version: 3] [Class: CrealityCXDLPFile]
```

### Printer and Slicer

- Printer: Creality Halot-One (CL-60)
- Slicer: n/a

### Description of the bug

After generating the PCB image, the image appears normally in the working area. However, after exporting the cxdlp file, the printer refuses to print it, with the error "There is a problem with this model data!"

[image]

### How to reproduce

1. Import a cxdlp file with a working print with File -> Open -> select working.cxdlp -> open
2. Go to Tools -> PCB Exposure -> Add -> import GBR
3. GBR opens on preview, no problems.
4. Click PCB Exposure
5. PCB appears on working area as per preview, no problems.
6. File -> Save as -> pov_disp.cxdlp
7. Copy file to usb drive and open on printer

### Files

[files.zip](https://github.com/user-attachments/files/20741163/files.zip)

sn4k3, 2025-06-15:
Can't find anything wrong with the file.
However I see it touch the bounds of screen, can you try to resize and give it some margin to see if thats the case of the problem?

felipetrentin, 2025-06-15:
Thanks for the quick reply! I spent some time trying with other files. I turns out touching the borders is not the issue, and is actually very simple. It's kinda embarrassing how i didn't think of this earlier.

to solve it, just go to
Tools -> edit print parameters
and set both lift heights to some non-zero value (I set it to 1mm)
apparently the printer has to lift in order to print.

Also, I know this isn't really the point of this issue, bot if the grb file doesn't have a drill/place origin (I use KiCAD) it doesn't generate the PCB layer, but also not give any errors, which is very frustrating. I would suggest adding a label saying "no origin found" or something like that...

To solve the original issue however, I would suggest that the tool don't change any print parameters, or at least prompt the user if they want to change print parameters.

sn4k3, 2025-06-15:
Nice find, I will make the tweaks. Thanks.