sn4k3/UVtools #1039: [Bug] Incorrect rendering of circular pads from G02/G03 arcs (missing circles and unintended connections)

Rob0xFF, 2025-06-23:
### System

```Text
UVtools v5.1.3 X64
Operative system: Darwin 24.5.0 Darwin Kernel Version 24.5.0: Tue Apr 22 19:54:26 PDT 2025; root:xnu-11417.121.6~2/RELEASE_ARM64_T8112 Arm64
Processor: Apple M2
Processor cores: 8
Memory RAM: 3.30 / 4.95 GB
Runtime: osx-x64
Framework: .NET 9.0.6
AvaloniaUI: 11.3.1
OpenCV: 4.10.0

Sreens, resolution, working area, usable area:
1: 3440 x 1440 @ 100% (Primary) (On this)
    WA: 3440 x 1330    UA: 3440 x 1330

Path:       /Applications/UVtools.app/Contents/MacOS/
Executable: /Applications/UVtools.app/Contents/MacOS/UVtools
Loaded file: antenna.pwsz [Version: 0] [Class: AnycubicZipFile]
```

### Printer and Slicer

- Printer: Anycubic M7 Pro
- Slicer: Anycubic Photon Workshop

### Description of the bug

UV Tools does not correctly render circular features defined using G02/G03 arc commands in Gerber files. Initially, arcs are completely missing or rendered incorrectly, and unintended straight-line connections appear between the arc endpoints. Only after explicitly converting arcs into segmented G01 lines and adding D02 moves between them does the rendering match the actual PCB layout.

### How to reproduce

Details

This issue affects Gerber files that use G02/G03 for circular pads (e.g. full circles defined with matching start and end points). The issue manifests in two stages:
	1.	Initial problem (most critical):
	•	UV Tools fails to display the circles at all, as if the arcs are ignored.
	•	It instead shows phantom straight lines between some arc start/end points.
	2.	After converting arcs to G01 segments:
	•	The circular shapes appear.
	•	However, straight-line connections between arcs still remain — even when these arcs are logically unrelated and should be isolated.
	3.	Final workaround (fixes everything):
	•	After inserting a D02 (move) command before each arc or circle, the rendering becomes correct.
	•	This suggests that UV Tools is implicitly connecting consecutive arc endpoints unless explicitly told to “lift the pen”.

Expected Behavior
	•	G02/G03 arcs should render correctly as curved geometry, including full circles.
	•	Arcs that are not part of a continuous interpolation (e.g. not within a G36/G37 polygon) should not be connected to each other unless explicitly specified.
	•	A missing D02 should not cause artificial connections between logically separate features.

Suggested Fix

Improve handling of G02/G03 commands by:
	•	Properly recognizing full-circle arcs defined with matching start/end points and valid I/J offsets.
	•	Avoiding implicit line connections between arcs unless they are within a G36/G37 block or part of a continuous draw.
	•	Optionally warning when interpretation is ambiguous due to missing D02.

How to Reproduce
	•	Load a Gerber file with multiple circular pads using G02/G03.
	•	Observe that circles are either missing or incorrectly connected.
	•	Apply post-processing:
	1.	Convert arcs to G01 segments (still shows wrong connections)
	2.	Add D02 move commands before each arc — only then does the rendering match the original design.

Versions Affected
	•	UV Tools 5.1.3
	•	Machine: Any
	•	Platform: macOS + Windows (tested)

Reference

KiCad GerbView and Gerbv both display the arcs correctly without any unintended connections or omissions.

### Files

G75*
%MOMM*%
%FSLAX34Y34*%
%ADD10C,0.150*%
G54D10*

G36*
X80000Y100000D02*
G03X80000Y100000I20000J0D01*
G37*

G36*
X180000Y100000D02*
G03X180000Y100000I20000J0D01*
G37*

G36*
X280000Y100000D02*
G03X280000Y100000I20000J0D01*
G37*
M02*

sn4k3, 2025-06-23:
I didnt know gerber allow to use G01/2/3 inlined with XY, always tought G was a single line command.
By looking at gerber format again I found a very tiny example and reference under "Deprecated syntax", in other words it should not be used anymore, no clue why recent cad issue such.
Also, using shapes that draw differently inside a region don't look right, they should be outside a region/polygon.

[image]

As it's part of gerber and kicad renders, I have made fixes to allow that syntax and now produces:

[image]

Can you provide more samples about the other problems you talked about?

sn4k3, 2025-06-28:
Fixed

closed 2025-06-28 by sn4k3

Rob0xFF, 2025-06-28:
Both problems fixed in latest release. Thank you very much!