sn4k3/UVtools #1044: Elegoo Saturn 4 Ultra16K - function Calibration/Tolerance - Abnormal CPU&RAM consumption - Crash at end with InsufficientMemoryException

JPJJPJ, 2025-07-10:
### System

```Text
UVtools v5.1.4 X64
Operative system: Microsoft Windows 10.0.19045 X64
Graphic card: 
Processor: Intel(R) Core(TM) i7-5500U CPU @ 2.40GHz
Processor cores: 4
Memory RAM: 10.00 / 15.90 GB
Runtime: win-x64
Framework: .NET 9.0.6
AvaloniaUI: 11.3.2
OpenCV: 4.10.0

Path:       C:\Program Files\UVtools\
Executable: C:\Program Files\UVtools\UVtools.exe
```

### Printer and Slicer

- Printer:  Elegoo Saturn Ultra 16K
- Slicer:  Elegoo SatelLite

### Description of the bug

With a goo file generated with Elegoo SatelLite, the tools Calibration/Tolerance consumes all the memory and the CPU to finally crash at the end when trying to "Generate Tolerance Test".

```
System.AggregateException: A Task's exception(s) were not observed either by Waiting on the Task or accessing its Exception property. As a result, the unobserved exception was rethrown by the finalizer thread. (Your system memory RAM hit the limit of 15.03GB from a total of 15.9GB.  Available: 0.87GB.  UVtools: 12.64GB (79.5%).
The program crashed on purpose due the impossibility to pause or cancel the running operation, this was to relief pressure and ensure the system stability.

Monitor your memory RAM under the system tools and re-run the operation to check the usages and find the cause.
You may need to stop other processes or increase your memory RAM in order to deal with huge files and/or heavy operations.

Currently you have configured a lower limit of 1GB of available memory RAM for the operations.)
 ---> System.InsufficientMemoryException: Your system memory RAM hit the limit of 15.03GB from a total of 15.9GB.  Available: 0.87GB.  UVtools: 12.64GB (79.5%).
The program crashed on purpose due the impossibility to pause or cancel the running operation, this was to relief pressure and ensure the system stability.

Monitor your memory RAM under the system tools and re-run the operation to check the usages and find the cause.
You may need to stop other processes or increase your memory RAM in order to deal with huge files and/or heavy operations.

Currently you have configured a lower limit of 1GB of available memory RAM for the operations.
   at UVtools.UI.MainWindow.<>c__DisplayClass57_0.<<RamUsageTimerOnElapsed>b__0>d.MoveNext() in UVtools.UI\MainWindow.axaml.cs:line 520
   --- End of inner exception stack trace ---
Category: Task
File: Cube_1_202507101153.goo  [Version: 0] [Class: GooFile]

Machine date time: 10/07/2025 12:04:26
       UTC date time: 10/07/2025 10:04:26
```

### How to reproduce

1 Load the goo file generated by Elegoo SatelLite software
   Around 5Go used
2 Launch Calibration / Tolerance => heavy CPU and RAM usage during 30sec
3 Tolerance Buton => heavy CPU and RAM usage during 90sec
4 Generate Tolerance test => very long => click a second time, display the dialog box and crash

JPJJPJ, 2025-07-10:
[Cube_1_202507101153.zip](https://github.com/user-attachments/files/21168113/Cube_1_202507101153.zip)
Sorry to have miss the attachement of the goo file.

JPJJPJ, 2025-07-17:
[image]

Under windows, I increased the virtual RAM file to 30 Go and the generation was successfull.
Could you tell me where is the ram limiter option? I did not find it in settings/General tab 

The process used 20Go and less than 2Go at end (see picture).
The ratio is very huge. Perhaps it is a memory management problem of the calibration tool.
What do you think?

Thank you !

sn4k3, 2025-07-20:
New release allow to pause operation, preventing app from kill in this situation.

closed 2025-07-20 by sn4k3

sn4k3, 2025-07-22:
1. Yes, all bitmaps are in memory but they are smart compressed and very efficient in space, example on this file 2893 14k layers, takes only 1249407821 bytes = **1.24 gb** in memory VS **197gb** if uncompressed. You can change compression in settings. Having them in memory is the most efficient way to get the job done, using disk to cache would slow everything and put a lot of stress in disk.

[image]

The problem comes when layer need to be used, and then it uncompress and get the real bitmap which are huge (14k) in this case. Multiplied with the concorrent tasks and the sub-mats it must create plus compress back it will generate a large heap pressure. After a loop it disposes the temporary objects but GC will release them only at GC pass or when required. In UVtools you can control the number of concurrent tasks and try to relief the problem, but the truth is you need tons of RAM to deal with these resolutions. For instance, if you open 2 x 14k bitmaps on photoshop you will see it takes more RAM than UVtools does in idle with it 2893 layers in memory.

2. There are no such thing as 1-bit bitmap, bitmaps comes as 8bit, 24bit and 32bit. Normally bitmaps are optimized for 24bit (RGB), UVtools uses 8bit maps. Lowest primitive type is a byte (8 bits). However is possible to do "smart compression" and use a byte to hold 8 pixels, but that is compression/RLE and will never be possible to use them like that, they must always be translated to a true bitmap to perform operations. That is what UVtools already do as explained in 1. plus UVtools crops the bitmap in memory, if your layer have 1 pixel, it will save just 1 pixel in memory, if 100x100 pixels it will compress 100x100 pixels and not the full 14K image. However when performing operations it must translate to full 14K again.

JPJJPJ, 2025-07-22:
Thank you for your answer.
My remarks are only to improve things if possible.
 
1- I am not sure to understand why the calibration tool needs all the layers in memory. It should release a layer after the compute of each one. 
2- Sorry to learn that the Standard C# bitmap object addresses 8bit. It is not optimal.

Do not worry about my PC: the SSD Disk is fantastic for swap file. 
Long but ok like printing !  ;-)

Regards

sn4k3, 2025-07-22:
1. They are, but as I explained, GC will release memory only when required and if it optimal. If you have a 200GB ram PC, GC will not care to clean 4GB of memory because you still have headroom. GC passes are expensive and is why you see programs consume ton of memory, inc: browsers, is not the real RAM they are using but the total RAM of use + ready to dispose. Windows does a "good" memory optimization, and more you have the more it will take and care less about defrag and optimize which occur more often when you are reaching the limit.
2. Not only C#, note UVtools uses OpenCV a high optimized C library, and it never had 1bit support. Modern libraries are default to 32 bits because CPUs are optimized to int's (32 bits) ends in faster operations. In fact you can't find 1bit bitmaps anymore, and some online services if you send an 8bit PNG, often outputs a 24/32 PNG. In the end it doesn't matter, OpenCV can't process 1bit maps and render engine to show layers require a 32bitmap.

sn4k3, 2025-07-23:
You can manage bits with bitwise operators, but the smallest type is byte (1 byte = 8bits) you don't have native bit type in C, even bool(0/1) is a byte/char = 8bits.
When you manipulate bits you are doing it under either a byte or a int, C# bitwise returns int's.
Formats such as CTB uses bits to compress in form for RLE, but even so there always must be a complete byte, even if data only contains a bit they must write the complete byte, this is because: "The smallest amount of data you can write at one time is a byte."
But again, it doesn't matter, UVtools already compress with LZ4 which is better than RLE bitwise, and to work with images it must be decoded to real bitmap without compression.