sn4k3/UVtools #1046: [Bug] clone layers, doesn't always increase layerheight if used on layer 1 and layer 0 is empty

djamu, 2025-07-17:
### System

```Text
UVtools v5.1.4 X64
Operative system: Microsoft Windows 6.1.7601 Service Pack 1 X64
Processor: Intel(R) Core(TM) i7 CPU         920  @ 2.67GHz
Processor cores: 8
Memory RAM: 9.20 / 12.00 GB
Runtime: win-x64
Framework: .NET 9.0.6
AvaloniaUI: 11.3.2
OpenCV: 4.10.0

Sreens, resolution, working area, usable area:
1: 2048 x 1152 @ 100%
    WA: 2048 x 1152    UA: 2048 x 1152
2: 2048 x 1152 @ 100% (Primary) (On this)
    WA: 2048 x 1152    UA: 2048 x 1152
3: 2048 x 1152 @ 100%
    WA: 2048 x 1152    UA: 2048 x 1152

Path:       C:\Program Files\UVtools\
Executable: C:\Program Files\UVtools\UVtools.exe
Loaded file: Impeller UTIII-R-63 B-printable-v3-20mm-thin.ctb [Version: 5] [Class: CTBEncryptedFile]
```

### Printer and Slicer

- Printer:  Elegoo Saturn4 Ultra 16K
- Slicer:  Lychee

### Description of the bug

When I use clone layers startying from the first layer, it increases the layer height eratically when layer 0 is empty.
I took 2screenshots to show what I did exactly. It doesn't add 0.3mm to the total height. The 2nd screenshot shows the situation working as intended.
Before this operation the layer heights are as follows.
This is originally a .goo file converted to .ctb V5

0 > 0.05mm ( empty layer )
1 > 0.05
2 > 0.1
3 > 0.15
4 > 0.2
5 > 0.25
6 > 0.3
....

[image]

after this operation layer heights are as follows
0 > 0.05mm
1 > 0.05
2 > 0.05  
3 > 0.1
4 > 0.1   *
5 > 0.15
6 > 0.2  
7 > 0.25
8 > 0.25  * 
9 > 0.3  ... 
rest of the layers is normal

if I insert layers from layer 2 everything is normal.

[image]

### How to reproduce

Bug is 100 % reproducable

djamu, 2025-07-17:
Maybe it's a good idea to add an automated "clone layers" to the suggestions settings with the option to auto apply. 
I print technical pieces straight on the bottom plate ( like an FDM ), rarely do I use supports. My printer and (probably many others) squashes the first layers a bit due to ( probably ) viscosity of the resin ( the build plate is located a bit higher then the printer thinks it is ). That means if the build plate goes up for example 0.05mm theplate actually sags a bit and 0.05 becomes for example 0.04...

If I want to print something that is exactly 1mm thick , it actually prints 0.7mm thicness ( something 10mm gets printed 9.7 ). To compensate for the squashing ( or sagging of the build plate ) I tend to clone 6 layers ( layers 2-7 ) to add 0.3 mm to the build.
Someone suggested it's schrinkage, but that is not true, printer layerheights don't schrink, steppermotor steps don't become smaller.

I know this is an edge case but I'm sure there are more people who encountered this issue

djamu, 2025-07-18:
I tested some things you mentioned.  (I'm a firmware developer, I specialized in high speed (+3000RPM) stepper motor control, I'll take a look at your firmware ).

How to reproduce:  

In suggestion settings > wait time before cure > disable Auto apply / disable "Create a first empty layer ..."  > save

[image]

I attached 3 small sliced files:

The original sliced file, not processed by UVtools
[test-pyramid-original-2mm-height.goo.zip](https://github.com/user-attachments/files/21315173/test-pyramid-original-2mm-height.goo.zip)

[image]

Cloning layer 1-6, 1 time adds the correct height  2mm  > 2.3mm
This gives the correct result

File processed by UVtools > added empty layer 

[test-pyramid-UVTools-added-empty-layer-2mm-height.goo.zip](https://github.com/user-attachments/files/21315238/test-pyramid-UVTools-added-empty-layer-2mm-height.goo.zip)

Cloning layer 1-6, 1 time does not give height of   2.3mm

This is the resulting file 

[test-pyramid-UVTools-added-empty-layer-wrongly-cloned-6-layers 1-6_2.15mm height should be 2.3.goo.zip](https://github.com/user-attachments/files/21315370/test-pyramid-UVTools-added-empty-layer-wrongly-cloned-6-layers.1-6_2.15mm.height.should.be.2.3.goo.zip)

My guess is that the clone layer function uses the difference in layer height, ( current - previous ) in case layer number > 0.
Since layer 0 is at the same height as layer 1. the increment is 0.    

In regard to this and the rest of your reply.

Increasing wait time is good advice 

> This is a natural problem of viscous/dense resins but mostly is a problem that companies refuse to solve, because it's actually easy. I solve this on my open-source firmware by pushing plate against lcd for sometime (Using TMC with low stall), then you still need to wait a bit for plate to reach the desired position. Its all a matter of getting the home correctly and wait for position to reach the desired/real position. If companies listen to users and not lock down mSLA we would be miles ahead. Instead they release tons of printers with same outdated and lacking firmwares.

I partly agree, viscosity plays a role. 
I installed a very accurate tilt sensor on the build plate, and if this was purely because of viscosity, it wouldn't happen without any resin present ( empty resin bath ).  But that's not the case. 
Build plates do sag and bend a bit and unless a printer has a symmetrical lifting mechanism ( pricey ) . Suction or pressure under the build plate caused by viscosity will just exacerbate the issue.
Your solution of waiting and pressing more ( I assume you press the plate down a bit more then it needs to be and then lift it up very slightly ? )  just mitigates the problem as it is a software solution ( a hack ) on what is essentially non precise hardware. ( I have no problem with that > CD players could use a plastic frame instead of a metal one the moment someone invented an autofocus algorithm / controller for the lense  : )

So I disagree with your satement that "companies refuse to solve it" as it is mostly a mechanical issue. Making hardware "unbendable" is expensive in materials.

A good alternative for that hardware would be to use a motionsensor on the build-plate (like I did now) and calculate the difference in angle...

sn4k3, 2025-07-18:
With your file I was able to reproduce, it will be fixed on next release.

>  I assume you press the plate down a bit more then it needs to be and then lift it up very slightly

Correct, with low current for easy stall and big wait times to let stabilize.
This problem is well documented and a reason why wait times suggestion surge in UVtools.

While the problem is mechanical and you may be right about the price for proper implementation, that alone doesn't guarantee the position if not willing to wait before exposure, even a 4 granity or symetrical axis may not be in right position to cure the layer if done in a rush. Sigle axis arm do bend and introduce bigger problem, but those can be "fixed" by software. The problem is that you don't have full control of your printer, unless you have a full gcode printer where you can issue the commands you want to do exactly what you want. 
You are with goo format and ctb format which allow per layer settings, but not every user have that "luck", anycubic, creality, and all others outside ctb, goo, and few others formats can't have per layer settings, so they are stuck to either wait for all layers or nothing at all. 
Even in formats that support per layer settings, tell me which software can make use of that? Only UVtools. No slicer make use of that, meaning that feature exists but is not used outside UVtools, which unfortunally means is poor/not supported, as you might know using such in goo format can brick your print, anycubic "supports" but if you use it the print just get abnormal with problems. In some firmware updates that flag causes problems where before they didn't.
This is what I mean by "companies refuse to solve it", years have past and the problems remains the same, they have good machines hardware wise, but firmware is bad and outdated, they refuse to listen and solve "minor" problems like this one. The moment they give you full control over printer with gcode not only you can solve this problem but implement a ton of new features. See what FDM can do firmware-wise and compare to mSLA. 

The moment I converted my printer to [klipper](https://github.com/Klipper3d/klipper/pull/6575) all my problems gone, but also have full web access and can implement whatever I want

- Horizontal lines due pause: Gone
- First layer problems: Gone
- First layer position due arm bent: Gone
- Exposure time precision: Fixed
- Proper configurable timelapse: Implemented
- Camera, filters, heaters as print start: Implemented
- Custom cicles and macros: Implemented
- Want to print supports and model in different exposure times? No problem!
- etc, etc...

All that goodies we can have but they refuse to implement gcode based printers to impose their crappy format and render your machine useless at some point. 
If firmwares followed same path as FDM (OpenSource) you could fix any problem.

PS: My firmware require a bit of tinkering not easy for the consumer to implement without a bit of experience with klipper and custom builds, should be easy for the FDM user who hacked the printer with custom parts and implement klipper, for example.

sn4k3, 2025-07-20:
Check last release, should be fixed.

closed 2025-07-21 by sn4k3