sn4k3/UVtools #1050: [Bug] UVTools does not see any file of any format

rhadeee, 2025-07-27:
### System

```Text
UVtools v5.1.5 X64
Operative system: Microsoft Windows 10.0.22631 X64
Processor: 12th Gen Intel(R) Core(TM) i7-12700
Processor cores: 20
Memory RAM: 10.15 / 63.79 GB
Runtime: win-x64
Framework: .NET 9.0.7
AvaloniaUI: 11.3.2
OpenCV: 4.10.0

Path:       F:\Program Files\UVtools\
Executable: F:\Program Files\UVtools\UVtools.exe
Loaded file: UVtools_demo_file.sl1s [Version: 0] [Class: SL1File]
```

### Printer and Slicer

- Printer: 
- Slicer: 

### Description of the bug

The software does not see any type of format when I try to open anything.  If I try to type in the name of the file I want they appear in the list but they don't open, I get sent to normal window like nothing happened.

### How to reproduce

1.Open UVTOOLS
2.Try to open any file, does not work, not even drag and drop

rhadeee, 2025-07-27:
> Can you post here the file names with extension you trying to open?

Hello. Nothin shows up, only .zip files apparently but formats like :
.chituubox , .goo , or even stl do not show up when using the open file
option.

sn4k3, 2025-07-27:
.chitubox, .stl are not valid slicer files for UVtools.
.CTB, .GOO should open. Try to drag and drop your goo file

rhadeee, 2025-07-29:
> Hmm, I will come back tomorrow with an update. It’s my first time trying
> to use it so I might have surely missed some important first use details.
> Should have started with that.

Yep, I was doing it all wrong, I apologize for that. Thanks for the quick
responses and sorry for my late replies :( Have a good night !