sn4k3/UVtools #1052: [Bug] Print lines in prints if saved from UVtools

deezdrama, 2025-08-04:
### System

```Text
UVtools v5.1.3 X64
Operative system: Microsoft Windows 10.0.26100 X64
Processor: AMD Ryzen 7 7800X3D 8-Core Processor           
Processor cores: 16
Memory RAM: 5.57 / 63.62 GB
Runtime: win-x64
Framework: .NET 9.0.6
AvaloniaUI: 11.3.1
OpenCV: 4.10.0

Sreens, resolution, working area, usable area:
1: 3840 x 2160 @ 200% (Primary) (On this)
    WA: 3840 x 2064    UA: 1920 x 1032

Path:       C:\Users\deezd\OneDrive\Desktop\UVtools_win-x64_v5.1.3\
Executable: C:\Users\deezd\OneDrive\Desktop\UVtools_win-x64_v5.1.3\UVtools.exe
Loaded file: UVtools_demo_file.sl1s [Version: 0] [Class: SL1File]
```

### Printer and Slicer

- Printer: Saturn 4 Ultra
- Slicer: Chitubox

### Description of the bug

Ive been using UVtools to find islands and resin traps after slicing in chitubox. I also add a 30 second rest before cure to base layers that transitions to 3 seconds for normal layers. This is all that I apply in UVtools.

In chitubox I already run a 3 sec rest before cure so no differences there, UVtools is basically just adding the 30 sec rest for base layers which wouldnt affect the overall print.

Ive printed 8 models now, 1/6 or 1/4 scale.... all of them exhibit bad and consistent print lines throughout. Ive ran a ton of exposure tests, Z axis tests, made sure Z axis rails were tight, re-greased z screw, etc....

I realized none of the exposure tests I printed had these lines so wondered if they were caused by UVtools, I loaded up a head from one of my recent models that had bad lines and sliced in chitu without running through UVtools afterwards and it printed flawlessly.

Some have wondered if UVtools was disabling any AA I had active in chitu, but I ran no AA on any of my models or tests. 

[image]

### How to reproduce

Any and all models I slice in chitu and run through UVtools has these print lines, whereas printed just from the chitu slice file is fine

### Files

[image]

[image]

[image]

[image]

sn4k3, 2025-08-04:
Need troubleshoot...
Try the following:

- Load model in UVtools
- Go pixel editor
- Remove a black pixel on a black pixel
- Save and print (Without any other modification)

Report back if same happens

deezdrama, 2025-08-05:
> > So basically you want me to just open this sliced file, do a quick pixel edit in UVtools and save/print?
> 
> Yes, to trigger a full encode without any modifications

Only loaded the print, added 1 pixel to top layer, saved and printed. Bad lines.

[image]

sn4k3, 2025-08-05:
Ok, send me both unedited and the last edited file

deezdrama, 2025-08-05:
> The file edited with UVtools turn a flag known to be problematic with most formats, CTB however wasn't many reports of issues with that, however try to disable this feature and repeat the print with the pixel trick:
> <img alt="Image" width="822" height="381" src="https://private-user-images.githubusercontent.com/113281/474288048-e41dd5a2-8c86-43c5-bfd9-12d2e565dabb.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTQzNjE1MjIsIm5iZiI6MTc1NDM2MTIyMiwicGF0aCI6Ii8xMTMyODEvNDc0Mjg4MDQ4LWU0MWRkNWEyLThjODYtNDNjNS1iZmQ5LTEyZDJlNTY1ZGFiYi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwODA1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDgwNVQwMjMzNDJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT05OTg4NjZhNDI4ZmRlOGIwMDI3YmRlMGZkOGQzNTJiN2FiYjAzNGUwYjA0ZmIzZThjMmNjNjg2YWQ0ZmE0MjZlJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.Y8bIuhFN-uWqvPbzfxYclmQoqbqRw8EhrsTLCPAGFJs">
> 
> After save file make sure this value is 0, as shown:
> <img alt="Image" width="415" height="918" src="https://private-user-images.githubusercontent.com/113281/474288260-416a5eb7-2801-4c2b-b33f-2a4f1091dc4c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTQzNjE1MjIsIm5iZiI6MTc1NDM2MTIyMiwicGF0aCI6Ii8xMTMyODEvNDc0Mjg4MjYwLTQxNmE1ZWI3LTI4MDEtNGMyYi1iMzNmLTJhNGYxMDkxZGM0Yy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwODA1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDgwNVQwMjMzNDJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT00MDkxYjZjYzZkZjliNmJjYmE1NzFhZTVmMTJkNjdiZThkZWZiOGU1OTcxODgxNWUyMDBmZGUxMDllN2U3MDA0JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.qa1JvW1tBlyKGu-R4z9mY3ahl5rETPRQwjfIbcPMlLc">

Awesome! That was fast troubleshooting!

I really appreciate it!
Ill give it a try tomorrow evening and report back

deezdrama, 2025-08-05:
Also I probably should of mentioned, sometimes when I try to open uvtools nothing happens and it takes the program 30seconds or more to actually open.
My PC gas good specs so always thought this was odd.
I wonder if I should of tried uninstalling and redownliading a fresh copy of uvtools?

Does the "per layer" setting usually work correctly for other saturn 4 ultra users? Why would it only cause issues for me?

Its been a while since I first installed UVtools, I cannot remember if I had to select a printer profile or if there were options I had to setup the first time I launched it for my printer. Is there any specific configuration needed for the saturn 4 ultra when first installing UVtools? Should I try re-installing?

Thanks

deezdrama, 2025-08-05:
> > Also I probably should of mentioned, sometimes when I try to open uvtools nothing happens and it takes the program 30seconds or more to actually open.
> > My PC has good specs so always thought this was odd.
> > I wonder if I should of tried uninstalling and redownliading a fresh copy of uvtools?
> 
> View under task manger to see what is happening, antivirus can stall programs to check things before run. Second run of program also takes much time? On my system it take about 3 seconds on first run and 1s after that.
> 
> You can try uninstall and install again but I suspect you will get same problem, however don't hurt to try. Settings are kept.
> 
> > Does the "per layer" setting usually work correctly for other saturn 4 ultra users? Why would it only cause issues for me?
> 
> Don't know, that feature was never used by slicers, printer firmware barely support it and when does it does more bad than good. Only UVtools use that flag to try "wilder" approach. It causes trouble for most of the users, and some which work after upgrade firmware to recent stop working. Did you try your print without the flag?
> <img alt="Image" width="333" height="269" src="https://private-user-images.githubusercontent.com/113281/474583558-ce13f876-d38d-461d-8371-6e3f1677a5b0.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTQ0MzIyMTYsIm5iZiI6MTc1NDQzMTkxNiwicGF0aCI6Ii8xMTMyODEvNDc0NTgzNTU4LWNlMTNmODc2LWQzOGQtNDYxZC04MzcxLTZlM2YxNjc3YTViMC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwODA1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDgwNVQyMjExNTZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1mNjYzMzI0YjcwYjdkMDk0OTM3MDJiYjJjMTAwMDc3OGYwNjVmZjU5OGZkMWQ0OWNiNGU3ZDA3MjhkZWU0ZTFmJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.pUFMIoq0L9A339uui2MrAGHSYXLl9lraUAOs7L4Sux8">
> 
> > ts been a while since I first installed UVtools, I cannot remember if I had to select a printer profile or if there were options I had to setup the first time I launched it for my printer. Is there any specific configuration needed for the saturn 4 ultra when first installing UVtools? Should I try re-installing?
> 
> No first time configs required or asked.

The per layer settings is already 0

[image]

sn4k3, 2025-08-05:
Yes, is 0 when come from slicer but turn into 64 after process with UVtools when setting is enabled.
Process file with UVtools and make sure to disable setting.

deezdrama, 2025-08-05:
Even though it was already reading 0 for per layer... I still went into settings and disabled per layer, I edited 1 pixel again and its printing now

deezdrama, 2025-08-06:
> Yes, is 0 when come from slicer but turn into 64 after process with UVtools when setting is enabled. Process file with UVtools and make sure to disable setting.

I disabled per layer setting, confirmed it was 0, edited 1 pixel again, saved and printed... Still bad print lines 😢

[image]

sn4k3, 2025-08-06:
You using latest UVtools version right? Because the version where you open issue was not the latest.
I notice some wierd parameters on your lifts due tilting vat, I fixed those values, try this file:

[test Head uv tools debug1.zip](https://github.com/user-attachments/files/21609391/test.Head.uv.tools.debug1.zip)

deezdrama, 2025-08-06:
> You using latest UVtools version right? Because the version where you open issue was not the latest. I notice some wierd parameters on your lifts due tilting vat, I fixed those values, try this [...]

Yep, I just installed the latest build for this test.
I thought lift settings wouldnt matter anyway because the printer would ignore them since its tilt vat?

deezdrama, 2025-08-06:
> > I thought lift settings wouldnt matter anyway because the printer would ignore them since its tilt vat?
> 
> Would be good but no, users report abnormal behaviour when using different values, that was patched some versions ago. Try my file and report back.

Its printing now, Ill check on it in the morning. Thanks

deezdrama, 2025-08-06:
> > I thought lift settings wouldnt matter anyway because the printer would ignore them since its tilt vat?
> 
> Would be good but no, users report abnormal behaviour when using different values, that was patched some versions ago. Try my file and report back.

Awesome, that seems to have fixed the line issue! 👍👍

So what was the lift setting that likely caused issues? Or what do I need to do to fix this on my end and to make sure it doesnt happen again?

Im curious also .... Say a lift setting is way off.... How does a tilting vat system like sauturn 4 u implement lift settings if it employs its own lift/tilt operations at the printer level? Basically asking what was going on mechanically to cause these visible lines?

Thanks again! Ill be sharing my findings tonight so it will hopefully help any saturn 4 u users that may run into issues.

I appreciate the help and your software!

sn4k3, 2025-08-06:
> So what was the lift setting that likely caused issues? 

The lift values were using high values, did you by chance change them in UVtools? Because they aren't auto changed.
As you were using a older version it might not include some patch, idk but the 5.1.3 should already constrain lift values for tinting vat printer.

> Or what do I need to do to fix this on my end and to make sure it doesn't happen again?

UVtools enforce values, you need to wait for the patch.

> Im curious also .... Say a lift setting is way off.... How does a tilting vat system like sauturn 4 u implement lift settings if it employs its own lift/tilt operations at the printer level? Basically asking what was going on mechanically to cause these visible lines?

I don't know, that are never explained to us and lifts are hidden from user in tilting vat. I observed that lifts are equal to layer height, so all lifts must be 0.05mm if you slice with 0.05mm. 
I think lifts also control the tilting speed, from normal to fast, but never catch the difference.
As there are no documentation nor information about these implementation we can only experimenting but as movements are minimal we can't conclude much without proper tools.

deezdrama, 2025-08-06:
> > So what was the lift setting that likely caused issues?
> 
> The lift values were using high values, did you by chance change them in UVtools? Because they aren't auto changed. As you were using a older version it might not include some patch, idk but the 5.1.3 should already constrain lift values for tinting vat printer.
> <img alt="Image" width="1587" height="665" src="https://private-user-images.githubusercontent.com/113281/475056520-6b51eacb-37b3-4a9f-8ed2-99db289d9e61.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTQ0OTM1MTksIm5iZiI6MTc1NDQ5MzIxOSwicGF0aCI6Ii8xMTMyODEvNDc1MDU2NTIwLTZiNTFlYWNiLTM3YjMtNGE5Zi04ZWQyLTk5ZGIyODlkOWU2MS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwODA2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDgwNlQxNTEzMzlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT00MDhjNGEzMmJjMzA4ZGNhMjZkNTk4YzAyZmQyZWQzNjc0YmZmNzFkNGZjNWI4NzE4MzU4MTE4ODFhNzhiZTIwJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.tmtVVUrsXUqsZlTWXxcsaxyRH7MMCdsA7IWrPrrygxM">
> 
> > Or what do I need to do to fix this on my end and to make sure it doesn't happen again?
> 
> UVtools enforce values, you need to wait for the patch.
> 
> > Im curious also .... Say a lift setting is way off.... How does a tilting vat system like sauturn 4 u implement lift settings if it employs its own lift/tilt operations at the printer level? Basically asking what was going on mechanically to cause these visible lines?
> 
> I don't know, that are never explained to us and lifts are hidden from user in tilting vat. I observed that lifts are equal to layer height, so all lifts must be 0.05mm if you slice with 0.05mm. I think lifts also control the tilting speed, from normal to fast, but never catch the difference. As there are no documentation nor information about these implementation we can only experimenting but as movements are minimal we can't conclude much without proper tools.
> 
> Can you send me your last processed file with the recent UVtools that failed, to check which values it written?

I never messed with any lift settings. All i ever adjusted was rest before cure delays for base and normal layers.

The last failed print was also on latest uvtools release, fresh install, nothing adjusted

sn4k3, 2025-08-06:
Ok, new version is up, you can try with it

deezdrama, 2025-08-06:
> Yes

Will try when i get home tonight. Thanks!

deezdrama, 2025-08-06:
This was the last file that failed that I tested...

https://drive.google.com/file/d/1j1tqLpFGtdKjX-k7f7Mzkm4y-Mf7lUP0/view?usp=sharing

Im getting ready to install new UVtools version, is there any specific lift range value I should keep an eye on that may indicate if its going to have the line issue?

sn4k3, 2025-08-06:
> Im getting ready to install new UVtools version, is there any specific lift range value I should keep an eye on that may indicate if its going to have the line issue?

it's fixed value now: 0.05mm should solve the problem

deezdrama, 2025-08-06:
> > Im getting ready to install new UVtools version, is there any specific lift range value I should keep an eye on that may indicate if its going to have the line issue?
> 
> it's fixed value now: 0.05mm should solve the problem

Awesome, quick question.... I loaded the test print, I added a custom rest time for base layers of 35sec and this transitions to 2.5 sec for normal layers. I can set 10 transition layers and apply but under info it always just says only 4 transition layers. This jumps from 35sec to 2.5sec for normal layers too fast. Any idea why its not adding the 10 transition layers I set?

sn4k3, 2025-08-07:
UVtools set the transition layers in two ways, it should work. Increase the value if you need more steps.

> Now that we know lift settings were the issue should i turn per layer settings back on?

You can try.

closed 2025-08-07 by deezdrama